g7uhn/ft817_buddy #3: Brief fluctuations in displayed data

drorthogonal, 2020-10-22:
The display occasionally updates with missing data or incorrect data, for just a few milli seconds, before reverting to the correct info. This is seen in two places while on the 2nd Page (speed change page).
1. Filter display switches from "CW N" to "???N".
2. Frequency display, displays random changes to the Mhz, KHz or 100s Hz or all three.

g7uhn, 2020-10-22:
Yes, these seem to be single read errors from the serial input, recovered on the next poll of the item.  I suspect the read errors are down to use of SoftwareSerial.  Note that the FT-817 serial interface should be using 2 stop bits (8N2) but SoftwareSerial can only do 8N1.  We seem to get away with just 1 stop bit but maybe this is the result.  I've also seen reports that SoftwareSerial can produce relatively frequent read errors... other software serial libraries are available.

Possible ways forward:
- Do nothing.  It's occasional at 9600 baud, probably tolerable.  But at 38400 baud the errors appear frequently;
- Use an alternate software serial library e.g. [NeoSWSerial](https://www.arduino.cc/reference/en/libraries/neoswserial/);
- Move to a microcontroller that has a dedicated hardware serial port for the radio interface.

g7uhn, 2020-11-07:
Moved to hardware serial on Rev2 - looks rock solid now.  :-)

closed 2020-11-07 by g7uhn