g7uhn/ft817_buddy #4: Toggle option for display backlight

drorthogonal, 2020-10-22:
#enhancement (trying to work out how to add tags!)
The short duration backlight on a normal press of the [light] button is the standard requirement and duration is good.
It would be nice to have the option to permanently toggle the backlight on and off, maybe using a [shift] and [light] combination or a long press on the [light] button.

g7uhn, 2020-11-07:
Good suggestion.
Also need to:
- improve the responsiveness of the LIGHT button to match the other buttons
- make backlight non-blocking

g7uhn, 2020-11-11:
- Temporary fix made to change it to a toggled backlight...
- Now non-blocking
- Button works the same as PAGE

Intent is to introduce a small counter that gets decremented on each pass through the main loop and that will be the 'countdown' for timed backlight (although the duration of the backlight would become linked to the duration of the main loop which depends on how many status items are being polled...)

Probably a better way of doing it exists.

g7uhn, 2020-11-11:
Done. 
Timing improved, reduced chance of "double-press".  Non-blocking operation.
LIGHT = Timed backlight (4 cycles through slow loop)
SHIFT+LIGHT = Backlight ON
A further press of LIGHT when the backlight is on turns it off.

closed 2020-11-11 by g7uhn