g7uhn/ft817_buddy #5: Keyer speed selection

drorthogonal, 2020-10-24:
The current keyer speed options (page 2) are set up at compile time and not configurable from the FT817 Buddy.
This issue is to investigate a possible evolution to this to provide some more flexibility. Options include:
1. Providing a way to configure  and store user preferences for the speeds (would need a way to store values persistently). Probably beyond the scope of this project.
2. Read the current Keyer speed and provide 6 x soft key options (3 x slower and 3 x faster) e.g. if current keyer speed is 18wpm provide softly options for, say, 12, 14, 16, 20, 22 & 24wpm.
3. Read and display the current Keyer speed settings and provide [up] [down] options on the soft keys to increment or decrement the speed in, say, 1wpm steps, reloading the display with the newly set speed.
#enhancement
...I'm adding this one as a possible development for me to try to learn C++!

g7uhn, 2020-11-07:
Good enhancement... might be solved by integration of a K1EL keyer onto the Buddy... let's wait and see how we feel when we try that.

g7uhn, 2020-11-17:
Including the K1EL K16 keyer chip... the CMD+paddle speed adjustment seems to hit points 2 and 3

g7uhn, 2020-11-21:
I think we're there for a first pass at this.

- K1EL external keyer added to Rev3, powers up at a preset favourite wpm and speed +/- is controlled by CMD+paddle, beeps for feedback, see [K16 manual](https://hamcrafters2.com/files/k16man_R10.pdf), page 15, "Fast Speed Change Feature"
- If using the FT-817 internal keyer (unlikely after the addition of K1EL) then preset favourite speeds can be programmed/ modified into Buddy code easily (current code is an example)
- More complex read/modify of the internal keyer is possible if someone wants to put the time in... :smile:

closed 2020-11-21 by g7uhn